coussej/node-opcua-logger #8: Different timestamps on monitored and polled

felixkretschmer, 2017-09-02:
As #7 different timestamps are probably used. Polled does not take the opcua timestamp but the one at moment of polling. Should be opcua timestamp which is supposed to be the moment the value was generated.

coussej, 2017-09-04:
Yes, this causes the difference. Monitored values use the UA timestamp, while the Timestamp for polled values is generated at the side of the logger (and neatly rounded at second level). The reason for this is that polled values at a certain moment will share the same timestamp, making it easier to work with afterwards.

closed 2017-09-04 by coussej

felixkretschmer, 2017-09-04:
Regarding data correctness and the event of the value the timestamp of the UA server should be the one used for polled values instead. Otherwise the event of a value is falsified

coussej, 2017-09-04:
I see your point. Would be happy to accept a PR that makes the source of the timestamp for polled nodes configurable!