coussej/node-opcua-logger #11: Difference between polled and monitored + function of deadband

psmt, 2018-02-22:
Hi, 

Nicely working application!  
I do have a question about the exact functionality.  I did some tests and my conclusion about the functionality of polled and monitored is as follows:
**Polled**:  

- works cyclic, as expected, logs xx samples/minute 

- deadband check:  deadbandvalues are taken into account, so logging is conditional if  deadbandvalues are used (<>0) 

**Monitored**: 

- works event driven, checks value every xx ms (monitorresolution)

- **deadband values do not seem to have influence,** unconditional logging, if opc value has changed, even the slightest amount,  so deadband values do not seem to have any influence.

Can someone confirm this?  If so, is this as foreseen?

kind regards
Pascal

coussej, 2018-02-22:
Hi Pascal, 

This behaviour is indeed exactly as it was implemented. Monitored values are usually used for looking at status changes, valve states, ... so the deadband didn't seem necessary. 

Do you have a special use case that requires other behaviour?

psmt, 2018-02-23:
Hi, 

Thanks for the response.  After your explanation it seems logic to me and I think I can manage. 
In the sample config file, the deadband parameters are also there for the "monitored" option.  Maybe that was the reason for the confusion from the start.

# A monitored node
[[measurements]]
name               = "Int32monitored"
tags               = { tag1 = "test", tag2 = "AB43" }
nodeId             = "ns=2;i=10849"
collectionType     = "monitored"
monitorResolution  = 1000    # ms 
deadbandAbsolute   = 0 		# Absolute max difference for a value not to be collected
deadbandRelative   = 0    	# Relative max difference for a value not to be collected

coussej, 2018-03-01:
@psmt You're right. That is confusing. I'll update the readme!

closed 2018-03-01 by coussej