coussej/node-opcua-logger #13: Nothing write into InfluxDB.

BigeYoung, 2019-03-01:
I am very sorry to bother you, but I can't solve this problem myself.

After running `node logger.js`, I got only one line output:
```bash
influx_1 : starting writepump [ writeLimit:  1000 , writeInterval: 3000 ].
```

also, and I can see `influxd` continuous output like following:
```bash
[httpd] 127.0.0.1 - - [01/Mar/2019:10:23:02 +0800] "POST /write?u=&p=&database=test&precision=ms&db=test HTTP/1.1" 204 0 "-" "-" f025f9ed-3bc8-11e9-8042-2c600c872352 100
[httpd] 127.0.0.1 - - [01/Mar/2019:10:23:05 +0800] "POST /write?u=&p=&database=test&precision=ms&db=test HTTP/1.1" 204 0 "-" "-" f1f11e43-3bc8-11e9-8043-2c600c872352 93
[httpd] 127.0.0.1 - - [01/Mar/2019:10:23:08 +0800] "POST /write?u=&p=&database=test&precision=ms&db=test HTTP/1.1" 204 0 "-" "-" f3bbe921-3bc8-11e9-8044-2c600c872352 69
[httpd] 127.0.0.1 - - [01/Mar/2019:10:23:11 +0800] "POST /write?u=&p=&database=test&precision=ms&db=test HTTP/1.1" 204 0 "-" "-" f5869f09-3bc8-11e9-8045-2c600c872352 167
[httpd] 127.0.0.1 - - [01/Mar/2019:10:23:14 +0800] "POST /write?u=&p=&database=test&precision=ms&db=test HTTP/1.1" 204 0 "-" "-" f75183ec-3bc8-11e9-8046-2c600c872352 166
```

It looks nomal, but actually I got nothing changed in my Influxed database, even the `SHOW MEASUREMENTS` found nothing.

In `wireshark`, I found those `POST /write` message followed by the words `content-length:0  Connection:close`.

...While I can see my value changed frequently in `opcua-client`.

### More Information
with InfluxDB v1.7.4.

The config file `config.toml` for `node-opcua-logger`:
```toml
[input]
url             = "opc.tcp://cMT-0489:4840/G02"
failoverTimeout = 5000 # time to wait before reconnection in case of failure

[output]
name             = "influx_1"
type             = "influxdb"
host             = "localhost"
port             = 8086
protocol         = "http"
username         = ""
password         = ""
database         = "test"
failoverTimeout  = 10000
bufferMaxSize    = 64
writeInterval    = 3000
writeMaxPoints   = 1000

# A polled node:
[[measurements]]
name               = "Omron_T048"
tags               = { tag1 = "test", tag2 = "T4028" }
nodeId             = "ns=2;s=OMRON.Tags.Timer.T4028"
collectionType     = "polled"
pollRate           = 60     # samples / minute.
deadbandAbsolute   = 0      # Absolute max difference for a value not to be collected
deadbandRelative   = 0.0    # Relative max difference for a value not to be collected

# A monitored node
[[measurements]]
name               = "Omron_T4030"
tags               = { tag1 = "test", tag2 = "T4030" }
nodeId             = "ns=2;s=OMRON.Tags.Timer.T4030"
collectionType     = "monitored"
monitorResolution  = 1000    # ms
deadbandAbsolute   = 0 		# Absolute max difference for a value not to be collected
deadbandRelative   = 0    	# Relative max difference for a value not to be collected
```

Thanks for your reading again.

BigeYoung, 2019-03-01:
Problem solved.
**Just DON'T excute `npm audit fix`**

closed 2019-03-01 by BigeYoung

sxbaumc1, 2019-07-30:
I just got stuck with the same problem - could you explain how you solved this ?
best regards