coussej/node-opcua-logger #16: How to set URI correctly

Hans85, 2019-04-27:
Hello,

i try to set up the logger correctly, here is my configuration:

`
[input]
url             = "opc.tcp://user:password@10.164.6.149:48010/UA"
failoverTimeout = 5000 # time to wait before reconnection in case of failure

[output]
name             = "influx_1"
type             = "influxdb"
host             = "172.18.0.2"
port             = 8086
protocol         = "http"
username         = ""
password         = ""
database         = "rh_linie_test"
failoverTimeout  = 10000
bufferMaxSize    = 64
writeInterval    = 3000
writeMaxPoints   = 1000

# A polled node:
#[[measurements]]
#name               = "Int16polled"
#tags               = { tag1 = "test", tag2 = "AB43" }
#tags               = { tag1 = "testtag" }
#nodeId             = "ns=4;s=RH Montage 2. Config.RH Master.GlobalVars.00DB ZEITEN_ZAEHLER.SPS_ZYKLUS_AKT"
#collectionType     = "polled"
#pollRate           = 60     # samples / minute.
#deadbandAbsolute   = 0      # Absolute max difference for a value not to be collected
#deadbandRelative   = 0.0    # Relative max difference for a value not to be collected

# A monitored node
[[measurements]]
name               = "Int32monitored"
dataType           = "number"
tags               = { tag1 = "test", tag2 = "AB43" }
nodeId             = "ns=4;s=RH Montage 2. Config.RH Master.GlobalVars.00DB ZEITEN_ZAEHLER.SPS_ZYKLUS_AKT"
collectionType     = "monitored"
monitorResolution  = 1000    # ms
deadbandAbsolute   = 0 		# Absolute max difference for a value not to be collected
deadbandRelative   = 0    	# Relative max difference for a value not to be collected
`

but all i get is the image in the attachment. It doesn't log to my influxdb.

Can you help me?

[image]

Hans85, 2019-04-27:
[image]

erossignon, 2019-04-30:
```
url             = "opc.tcp://10.164.6.149:48010" 
```
try this instead:
```
url             = "opc.tcp://m01009186:48010" 
```

coussej, 2019-05-23:
Hi @Hans85, 

sorry for the late reply, I've been away on holiday..

In the second case, you have successfully connected to the OPCUA server socket, but the OPCUA client library only allows connection if the URI you specified is exactly the same as the endpoints exposed by the OPCUA server. That is why you get the error `end point must exist`.

The solution should be like @erossignon already mentioned. Have you tried that?

If you cant connect using the server name (if you have no DNS resolution for example), you either add an endpoint on the server that binds on the IP, or you can add an entry to the clients /etc/hosts file that maps the server name to the IP.

Hope that helps!

mahadi, 2019-08-02:
I also have the problem that i am not receiving any value. Using the UAExpert, the values are working fine, so something is left to my config that breaks it... is there any way to output some debug information?

coussej, 2019-10-30:
closing due inactivity. 

@mahadi if you have an issue that is different from this one, feel free to open a seperate ticket.

closed 2019-10-30 by coussej