coussej/node-opcua-logger #17: FATAL ERROR: CALL_AND_RETRY_LAST Allocation failed - JavaScript heap out of memory

alexmc1510, 2019-05-23:
Hello coussej,

I am testing your code "at limits" (more than 300 tags) but after a while, I receive the following error. It seems to be related with the total memory used by "opclogger.db".
In the moment of this error, this file had 198.019kb which is a lot of memory.

Could you have a look? Thank you a lot.

`
    at C:\Program Files\node-opcua-logger-master\node_modules\influx\index.js:67:23
    at InfluxRequest._parseCallback (C:\Program Files\node-opcua-logger-master\node_modules\influx\lib\InfluxRequest.js:122:10)
    at Request._callback (C:\Program Files\node-opcua-logger-master\node_modules\influx\lib\InfluxRequest.js:111:10)
    at Request.self.callback (C:\Program Files\node-opcua-logger-master\node_modules\request\request.js:185:22)
    at Request.emit (events.js:189:13)
    at Request.<anonymous> (C:\Program Files\node-opcua-logger-master\node_modules\request\request.js:1161:10)
    at Request.emit (events.js:189:13)
    at IncomingMessage.<anonymous> (C:\Program Files\node-opcua-logger-master\node_modules\request\request.js:1083:12)
    at Object.onceWrapper (events.js:277:13)
    at IncomingMessage.emit (events.js:194:15)

<--- Last few GCs --->

[34036:00000148C03AD2C0]  1206855 ms: Mark-sweep 1008.2 (1067.6) -> 1008.2 (1034.6) MB, 746.0 / 0.0 ms  (average mu = 0.476, current mu = 0.000) last resort GC in old space requested
[34036:00000148C03AD2C0]  1207602 ms: Mark-sweep 1008.2 (1034.6) -> 1008.2 (1034.6) MB, 747.0 / 0.0 ms  (average mu = 0.306, current mu = 0.000) last resort GC in old space requested

<--- JS stacktrace --->

==== JS stack trace =========================================

    0: ExitFrame [pc: 0000024CF41079E0]
Security context: 0x01975e19e6e9 <JSObject>
    1: byteLength(aka byteLength) [000002C1B21F80D9] [buffer.js:~509] [pc=0000024CF55BD137](this=0x002ecd2026f1 <undefined>,string=0x007524a2e0f9 <Very long string[202771045]>,encoding=0x01975e1be311 <String[4]: utf8>)
    2: arguments adaptor frame: 3->2
    3: from [000002394879E8E9] [buffer.js:~199] [pc=0000024CF55C85CF](this=0x02c1b21d2b29 <JSFunction ...

FATAL ERROR: CALL_AND_RETRY_LAST Allocation failed - JavaScript heap out of memory
 1: 00007FF77538F04A v8::internal::GCIdleTimeHandler::GCIdleTimeHandler+5114
 2: 00007FF77536A0C6 node::MakeCallback+4518
 3: 00007FF77536AA30 node_module_register+2032
 4: 00007FF7755F20EE v8::internal::FatalProcessOutOfMemory+846
 5: 00007FF7755F201F v8::internal::FatalProcessOutOfMemory+639
 6: 00007FF775B12BC4 v8::internal::Heap::MaxHeapGrowingFactor+9556
 7: 00007FF775B110CB v8::internal::Heap::MaxHeapGrowingFactor+2651
 8: 00007FF7756CCA3B v8::internal::Factory::AllocateRawWithImmortalMap+59
 9: 00007FF7756CF4ED v8::internal::Factory::NewRawTwoByteString+77
10: 00007FF7758D9AA8 v8::internal::Smi::SmiPrint+536
11: 00007FF7755E582B v8::internal::StringHasher::UpdateIndex+219
12: 00007FF77560AB26 v8::String::Utf8Length+22
13: 00007FF77535370C node::Buffer::New+4332
14: 0000024CF41079E0

C:\Program Files\node-opcua-logger-master>
`

coussej, 2019-05-23:
Hi there! 

300 tags should not be 'at limits' 😄we've had cases with 1000s that had no issues. Unless you are running on limited hardware? What are the specs of the system?

May I also ask you which versions of NodeJS and specifically the influx library you are using?

alexmc1510, 2019-05-24:
Thank you for your answer. More than 1000??? wow....I promise I will try!!

My hardware specs are:
Intel Xeon CPU E3-1225 v5 @ 3.30Ghz
RAM 8,00 GB
Windows 10 64b

Node js version 10.15.3

Influxdb library: How can I check this?

I think is something related to opclogger.db wich doesn't stop growing and growing.

I also have 2 questions....regarding functionality:
- If I have 10 machines with 200variables each, do you recommend to use only one instance of opclogger and one influxdb o do you recommend to create one instance per machine.
- I understand poll and suscribe but, regarding your experience, which method do you recommend?

I have also developed an excel macro to convert Kepware csv export into "config.toml" format. If you are interested I can share it with you. It was a pain to write one by one.

Please, tell me if you need some help.

coussej, 2019-05-24:
You can check the influx lib version in `node_modules/influx/package.json`. 
Can you also post the settings of your config.toml file?

Another question, are you getting any data written to influx? The buffer data should only grow if the logger is not able to write anything to the database, and thus starts buffering. 

(we have experienced some issues with the current implementation of buffering, but I hope to trickle down some of the improvements of our commercial version back to the OS one soon. In either case, they should not pose an issue on this scale.)

alexmc1510, 2019-05-24:
Hello,

influxdb package v4.1.0

[config.txt](https://github.com/coussej/node-opcua-logger/files/3216171/config.txt)

Regarding influxdb is has registered data until it failed. What I could see now is that Chronograf was freezed, also grafana.....after influxdb restart, they started working. Could be something related influxdb?

Thank you!!

alexmc1510, 2019-05-27:
Hello,

I have reproduced the error and, before out of memory issue I receive this error:

opclogger Error: {"error":"partial write: unable to parse 'PC_SPC.actPartProgram,equipment=IL_PAR01,opcstatus=Good value=\" M169 ;pump reset\r\n G4F2\r\n /M170 ;pump on if not skip block\r\nEndif\r\n\r\n$MC_ORI_IPO_WITH_G_CODE=0\r\nNEWCONF\r\n\r\nM150H4 ;OEE integer DB210\r\n\r\nLog_Action(\"WJet part program started = \"\u003c\u003c$P_PROG[0])\r\n\r\nM17\r\n\" 1558951473842': bad timestamp dropped=0"}

The value of this tag is a very long string.

coussej, 2019-05-31:
Hi alex, it seems that influx is not able to parse the incoming points. Probably something goes wrong with encoding/decoding to the influx line protocol because of the value of your string tag.

Can you just check and see if the logging works as expected without including this tag, and deleting the buffer database? This version of the logger keep retrying to send the data to influx when it fails, but in this case this means an endless loop while the buffer database keeps growing, hence the memory error.