coussej/node-opcua-logger #18: OpenSSL Issue

AWanderingEngineer, 2019-05-30:
Hi,

First I want to say I'm pretty new at linux and so I apologize in advance is this is a minor issue with an obvious solution. I'm trying to get this package installed and upon running npm install from the node-opcua-logger directory I get several errors having to do with the creation of private keys and certificates using openSSL. I've attached a text file with all the errors (there were several, all related to openSSL). However they all appear to be duplicates of this:

```
###########OPENSSL ERROR ############
CWD = /home/bmcadmin/node-opcua-logger/node_modules/node-opcua-client
Command failed: "openssl" genrsa  -out "/home/bmcadmin/node-opcua-logger/node_modules/node-opcua-client/certificates/PKI/own/private/private_key.pem" -rand "/home/bmcadmin/node-opcua-logger/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd" 2048
Can't load /home/bmcadmin/node-opcua-logger/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd into RNG
139806547309696:error:2406F079:random number generator:RAND_load_file:Cannot open file:../crypto/rand/randfile.c:98:Filename=/home/bmcadmin/node-opcua-logger/node_modules/node-opcua-client/certificates/PKI/own/private/random.rnd
[OPCUA-Logger Errors.txt](https://github.com/coussej/node-opcua-logger/files/3238905/OPCUA-Logger.Errors.txt)
```
I'd really appreciate any help in resolving this issue. Thank!

coussej, 2019-05-31:
Hi, can you specify some details about your environment?

- OS?
- Node version
- Is openssl installed correctly? Which version?

Heli4ce, 2019-06-24:
I have the same issues, both unexperienced and "Can't load /home....../random.rnd into RNG":
My versions are:

user@computer:\~/node-opcua-logger$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.2 LTS
Release:        18.04
Codename:       bionic
user@computer:\~/node-opcua-logger$ openssl version
OpenSSL 1.1.1  11 Sep 2018
user@computer:\~/node-opcua-logger$ node --version
v8.10.0

I don't know how to tell if openssl is correctly installed...

Thanks!

coussej, 2019-10-25:
Hi @AWanderingEngineer, 

certificate generation is done by the underlying node OPCUA library, and I've had problems with it before. Could you try this with the new version (v2) of the logger, which also contains a much more recent version of the node-OPCUA library? Even better, if you download a binary from the release page, you don't need to generate anything yourself.