coussej/node-opcua-logger #21: Logger startup error

marcoweiland, 2019-08-21:
Hi,
if I try to start the logger I get the following error:

C:\node-opcua-logger-master>node logger.js
C:\node-opcua-logger-master\writepump.js:18
        this.output = new influx({
                      ^

TypeError: influx is not a constructor
    at new WritePump (C:\node-opcua-logger-master\writepump.js:18:16)
    at Object.<anonymous> (C:\node-opcua-logger-master\logger.js:11:10)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:831:12)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:622:3)

marcoweiland, 2019-08-21:
forgot to mention:
OS: Windows Server 2016 Standard
NodeJS: 10.16.3 , tried both 64 and 32bit, same result

coussej, 2019-08-23:
Hello there, did you run `npm install` succesfully after cloning?

marcoweiland, 2019-08-25:
Yes, and everything seems to be fine

coussej, 2019-10-25:
Hi Marco! I've just merged v2 into master, containing lots of fixes and package updates. Could you check it out and see if it works for you?

coussej, 2020-03-26:
closing as stale

closed 2020-03-26 by coussej