coussej/node-opcua-logger #23: Installation v2

KoenWindey, 2019-09-28:
Hi Jeroen,

I've been trying to install the logger on Ubuntu 18.0.4. First I tried the master, but got all kinds of issues because it seems to refer to an old node opcua version, which throws errors on Ubuntu. 

Then I noticed the v2 branch and that seems to have a more recent version of node opcua - so I tried the install for that. That worked: no errors during 'npm install'.

However, as soon as I ran 'node logger.js' I got an error: Error: Cannot find module 'nedb'. I resolved this by running  `npm install nedb --save`. Perhaps it needs to be added to your dependencies in order to avoid this in the future.

Next issue, again as soon as I run 'node logger.js', I get the following error

```
/home/koen/node-opcua-logger-2/writepump.js:18
	this.output = new influx({
	              ^

TypeError: influx is not a constructor
    at new WritePump (/home/koen/node-opcua-logger-2/writepump.js:18:16)
    at Object.<anonymous> (/home/koen/node-opcua-logger-2/logger.js:11:10)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:831:12)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:622:3)
```
I am not sure what to do about this. I do have influxdb installed (version 1.7.8).

Any suggestions?
Koen

coussej, 2019-10-17:
Hi Koen, 

yeah sorry, the v2 branch is stil heavily WIP. You should run `node ./src/index.js` instead of `node logger.js`. I still need to cleanup some files from the old version, planning on releasing an alpha later this week.

coussej, 2019-10-25:
Hello Koen, 
v2 is now merged in master, and includes better documentation in the readme. 

Could you either try again from source, or download a binary from the release page? Let me know if it works!

closed 2020-05-27 by coussej