coussej/node-opcua-logger #26: Best Architecture

alexmc1510, 2019-10-27:
Hello and thanks in advance,
I would like to know which architecture do you recommends in my case. I have several machines which belongs same OPC server. 1. Do you recommend one instance of your software per machine or only one?
2. Do you recommend one influxdb per machine or only one?

I would like to know advantages/disadvantages (or incompatibilities)

Thank you in advance.

coussej, 2019-10-30:
Hi Alex, 
do you mean that you have several physical equipments, that are connected with the same OPC server? In that case I would just use one instance, installed on the OPC server, unless each machine has a very high number of metrics. 

I would also just use one influxDB. You can prefix all the metrics with a machine_id. 

If you want to log to more than 1 database within the same influxdb server, you'll have to use multiple instances of the logger.

alexmc1510, 2019-11-27:
Understood.
Thank you!

closed 2019-11-27 by alexmc1510