coussej/node-opcua-logger #30: [WARN] buffer - MaxBatchSize Exceeded

FGRav, 2019-11-21:
Hi there, 
thanks for sharing the script.
I am logging almost 2000 tags right now.

Couple of questions:
1. Why I am getting this warning? and how can i fix it?
[WARN] buffer - MaxBatchSize Exceeded

coussej, 2019-12-03:
The warning message is there for debug purposes, but it's probably better to put this at the DEBUG level so your logs dont fill up with the message.

Environment variables are set on the system environment running the actual process: https://en.wikipedia.org/wiki/Environment_variable 

In windows, you can use the command `set var1=value1` to set them, in unix `export var1=value1'. When you run the logger though a service daemon like systemd you can also configure the environment variables specifically for that process there.

alexmc1510, 2021-02-22:
Hello, I have faced the same issue. As I am using binaries, what environment variable should I change? I am not able to find the correct one. Actually running on Windows Server 2016.