coussej/node-opcua-logger #34: support for deadband in v2

scelle, 2020-01-28:
Hi, 

Are there plans for adding deadband facilities, similar to the first version of the logger?

kind regards
Pascal

coussej, 2020-01-30:
Hi Pascal, 

I removed that functionality because I thought it was not usefull, as influx has such good compression that storage shouldn't really be an issue. 

Could you perhaps elaborate on your use-case?

scelle, 2020-01-31:
If you have for instance a temperature sensor with an accuracy of +- 1% or +-0.5°C and you have a process where in 99% of the time the temperature is constant.  You only want to log the changes caused by the process, not the small fluctuations that are caused by a sensor inaccuracy.
This would  make a whole lot of difference in data traffic, storage, query performance, hardware etc, Certainly if you have xx sensors on xx machines.
It simply seems like good practice to avoid "useless" data.  
Maybe oldschool thoughts? ;-)

lifeisawavesorideit, 2020-01-31:
Hi guys,

first of all I would like to thank @coussej for puttin in the effort and creating this logger.

On this point I must agree with @scelle, it would be interesting to have an option to save only the relevant data by that I mean high value changes not caused by sensor inaccuracy. Certainly this days storage costs almost nothing but it would be ideal to use less insead of using more.

samborambo, 2020-12-22:
I suggest that adding deadband isn't the best solution. Your sensors might have an absolute accuracy of +/-1% but the relative accuracy is useful for sub-sampling. That 'noise' in between your useful measurement periods can be down-sampled to give much better accuracy of the useful data. You're better off pouring the raw data into a short term bucket and running a continuous query in Flux to down-sample the readings into a useful set.