coussej/node-opcua-logger #36: Test on rpi.

twoway, 2020-02-05:
Hi
Referring to my comment on this post:  https://coussej.github.io/2019/10/28/Open-Source-Monitoring-of-Industrial-Systems-with-InfluxDB-using-OPC-UA-Part-2/

I have done the first tests on the Pi.
When i start with `npm run start` i get this in terminal

```
> influx-opcua-logger@2.0.0-alpha start /home/pi/node-opcua-logger
> node ./src/index.js

/home/pi/node-opcua-logger/node_modules/toml/lib/compiler.js:14
    for (var i = 0; i < nodes.length; i++) {
                              ^

TypeError: Cannot read property 'length' of undefined
    at reduce (/home/pi/node-opcua-logger/node_modules/toml/lib/compiler.js:14:31)
    at Object.compile (/home/pi/node-opcua-logger/node_modules/toml/lib/compiler.js:10:10)
    at Object.parse (/home/pi/node-opcua-logger/node_modules/toml/index.js:7:21)
    at Object.load (/home/pi/node-opcua-logger/src/config.js:14:17)
    at Object.<anonymous> (/home/pi/node-opcua-logger/src/index.js:8:19)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! influx-opcua-logger@2.0.0-alpha start: `node ./src/index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the influx-opcua-logger@2.0.0-alpha start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-02-05T19_14_10_342Z-debug.log
```

The debug log:
```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'run', 'start' ]
2 info using npm@6.13.4
3 info using node@v10.18.1
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle influx-opcua-logger@2.0.0-alpha~prestart: influx-opcua-logger@2.0.0-alpha
6 info lifecycle influx-opcua-logger@2.0.0-alpha~start: influx-opcua-logger@2.0.0-alpha
7 verbose lifecycle influx-opcua-logger@2.0.0-alpha~start: unsafe-perm in lifecycle true
8 verbose lifecycle influx-opcua-logger@2.0.0-alpha~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/node-opcua-logger/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle influx-opcua-logger@2.0.0-alpha~start: CWD: /home/pi/node-opcua-logger
10 silly lifecycle influx-opcua-logger@2.0.0-alpha~start: Args: [ '-c', 'node ./src/index.js' ]
11 silly lifecycle influx-opcua-logger@2.0.0-alpha~start: Returned: code: 1  signal: null
12 info lifecycle influx-opcua-logger@2.0.0-alpha~start: Failed to exec start script
13 verbose stack Error: influx-opcua-logger@2.0.0-alpha start: `node ./src/index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid influx-opcua-logger@2.0.0-alpha
15 verbose cwd /home/pi/node-opcua-logger
16 verbose Linux 4.19.66-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "run" "start"
18 verbose node v10.18.1
19 verbose npm  v6.13.4
20 error code ELIFECYCLE
21 error errno 1
22 error influx-opcua-logger@2.0.0-alpha start: `node ./src/index.js`
22 error Exit status 1
23 error Failed at the influx-opcua-logger@2.0.0-alpha start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

twoway, 2020-02-05:
I followed this:

**From source**
- Install the latest version of NodeJS v10 on you system.
- Clone this repository.
- Run npm install in the project root.
- Create a config.toml of config.json file (see configuration).
- Run npm run start.
- Data!

coussej, 2020-02-13:
`/home/pi/node-opcua-logger/node_modules/toml/lib/compiler.js:14`

This looks like an error with your config file. Could you share it?

twoway, 2020-02-13:
Config file attached.
[config_toml.txt](https://github.com/coussej/node-opcua-logger/files/4200521/config_toml.txt)

twoway, 2020-02-13:
Config file attached.
[config_toml.txt](https://github.com/coussej/node-opcua-logger/files/4200533/config_toml.txt)

twoway, 2020-02-16:
I did a test with a .json config file. Looks better.
But i dont get a connection to UPCUA server because of cert. (Device=Beckhoff PLC )
The server is set to not use cert i think
I can connect with UA Expert only using uname and pass.
Can i disable the use of cert, or do you have any other solution?

```
pi@raspberrypi:~/node-opcua-logger $ npm run start

> influx-opcua-logger@2.0.0-alpha start /home/pi/node-opcua-logger
> node ./src/index.js

[2020-02-16T15:13:18.317] [INFO] main - Starting Influx OPCUA Logger v2.0.0-alpha, brought to you by FACTRY (www.factry.io)
[2020-02-16T15:13:18.333] [INFO] main - Initialising influxClient
[2020-02-16T15:13:18.497] [INFO] influx - localhost:8086 responded in 126ms running 1.7.9
[2020-02-16T15:13:18.499] [INFO] main - Initialising buffer
[2020-02-16T15:13:18.513] [INFO] main - Connecting OPCUA
[2020-02-16T15:13:25.668] [FATAL] main - Error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (getaddrinfo ENOTFOUND cx-4476c4)
    at Object.start (/home/pi/node-opcua-logger/src/opcua.js:60:11)
[2020-02-16T15:13:25.675] [INFO] buffer - Persisting membuffer.
```

coussej, 2020-02-18:
* Can you check the settings of the server? There are 3 auth modes:
anonymous, username/pass and certificate based. Which ones are enabled?
* If certificates are enabled, there should be a list of not-yet-trusted
certificates (from previous connection attemps) in de server settings that
you can trust.
* Did you try username/pass also in the config file?

twoway, 2020-02-19:
Have disabled auth. Same message. 
I log in ok without auth (anonymus & no cert) using UA Expert. 
I can try connecting to another server. Dou you have tips of a lihtweight sample server running on linux i can try?

coussej, 2020-02-20:
Looking back, it probably has nothing to do with auth. You get error
"getaddrinfo
ENOTFOUND" which means the host name could not be resolved. Can you ping
the PLC with the exact same hostname you used in the config file?

twoway, 2020-02-20:
Hmm
Assumptions it the mother of a f**ups...
I coiped the host name of PLC from UA expert to the Pi. Assuming it would work. My bad...
When I replaced with IP the logger started up!
It seems to work now...

coussej, 2020-02-22:
Great!

closed 2020-03-26 by coussej

twoway, 2020-07-17:
Hi.
I am experiencing crashes...
This is the error message.
Any tips?
[image]