coussej/node-opcua-logger #37: ANONYMOUS user token policy Error

DavidWicks, 2020-02-10:
Hi, I have been involved with PLCs and automation for many years, however I am very new to Nodejs, Git and have very limited linux experience.

I have managed to get the opc logger working in the demo mode succesfuly sending data to influx, however when I try to connect to OPC UA server I am unsuccesful.

OS: Windows 10 Pro 64 bit
Node v2.0.0
OPCUA Server:  KepServerEX 6.8

I have set the certificate under trusted clients in KepserverEX (first i had to convert it to der).

Any help you could provide would be appreciated.

[Annonymous user token polic error.txt](https://github.com/coussej/node-opcua-logger/files/4178543/Annonymous.user.token.polic.error.txt)
[config_toml.txt](https://github.com/coussej/node-opcua-logger/files/4178562/config_toml.txt)

coussej, 2020-02-10:
Hi here, 

it means you're trying to connect without a username and password (anonymous), but that anonymous access is not allowed by Kepware. You can enable it under the kepware project properties (and not in the UA configuration manager): See https://www.kepware.com/getattachment/de80e240-765e-451a-afce-640d413891c3/opc-ua-configuration-manager-manual.pdf, page 5/6: project properties > OPC UA

Either

DavidWicks, 2020-02-10:
Working !
Thankyou :)

closed 2020-02-10 by DavidWicks