coussej/node-opcua-logger #38: Metrics Configuration

DavidWicks, 2020-02-10:
Is there any examples/documentation of Metrics configurations ?
I am having problems understanding this, in particular the nodeId.
Would you be able to provde the correct configutation details for this example simulation in KepEX
Do I need to create an Alias first ?

[image]

[image]

coussej, 2020-02-12:
Hi David, 

Kepware exposes everything in namespace 2. After that, the nodeId is just the same path as you can see with the OPC quick client (accessible from the admin). For your examples, the nodeId's would be:

```
ns=2;s=Simulation Examples.Functions.Random2 (not sure about the space though)
ns=2;s=Channel1.Device1.Tag1
```

Does that make sense?

DavidWicks, 2020-02-13:
Thank you for the reply :)  

Yes it does make sense.  I am finding that there does not seem to be any solid reference data for the format of NodeID. (some conflicting info as well)

I confirmed with Kepware that the NodeID for the KepserverEx is fixed at 2.

The space between **_Simulation_** and **_Examples_** is valid.

I am now succesfully receiving simulation data into InfluxDB.

closed 2020-02-13 by DavidWicks