coussej/node-opcua-logger #41: Certificate problem

anantmk, 2020-02-28:
Hi Jeroen,

I am trying to test your opcua-logger. After **npm run start** I get an error because of no certificate. 

I have a machine with no internet access. Could you help me with this?

Thanks, Anant

Please see the logs:

[image]

[image]

coussej, 2020-03-02:
Hello,

did you run npm install on that same machine, or did you run it on a
different machine and then copied the folder?

In you case, the best thing to do is download a prebuilt binary. That
contains everything you need in a singe binary, and should run without
issues.

anantmk, 2020-03-04:
Thank you. The binary is working well!

Br, Anant

closed 2020-03-04 by anantmk