coussej/node-opcua-logger #43: How to connect multiple devices and containerization

mazamsezer, 2020-03-30:
Hello,
 I have two questions:

Firstly; 

Can you tell me how can I connect two or more plcs with your software, please?

Let's say I have two devices:
opc.tcp://bla.bla.bla.1
opc.tcp://bla.bla.bla.2

How a config file and measurements look like for this setup?

Secondly,

How can I run your software in Docker?

Thank you,

coussej, 2020-03-31:
-> For two devices, simply run the application 2 times with a different config file.

-> For docker, there are multiple ways, but I'd suggest a basic linux imagein which you curl the pre-built binary (release tab) and run it

mazamsezer, 2020-04-01:
Would you mind explaining what is curling in the pre-built binary in basic linux image a bit more, please? How I will do it?

mazamsezer, 2020-04-20:
I could manage to run this logger in a container. As I still don't know how to run more than one instance in a container with different config.toml files, I cloned my conrainer and run other config.toml through it.
It works like a charm!

I want to ask if and how can I connect OPC-DA clients to this logger.

coussej, 2020-04-21:
If you want to connect to an OPC-DA server, you'll need software for the DA to UA translation. You can use for examples Kepware KepserverEX or TaniOPC.

mazamsezer, 2020-06-08:
Thank you for your help.

closed 2020-06-08 by mazamsezer