coussej/node-opcua-logger #44: How to connect to influxdb cloud database

JanMaasland, 2020-04-16:
Hello,

I am new to influxdb and try to connect a local OPC UA server to my influxdb cloud account. I managed to connect to the OPC UA server, but have no idea on what the url should be.

Can you help me on this?

Regards,

Jan Maasland

coussej, 2020-04-17:
Hello, it should be in the form of http(s)://user:password@host:port/database

So, host should be the IP or FQDN of your cloud server, port is usually 8086, and the other details depend on how you confgres influxdb :-)

does that help?

JanMaasland, 2020-04-17:
I now get a connection to influx, but get the below message every 5 seconds. I assume that for the database part of the url I need to put my bucket?

[2020-04-17T14:58:57.652] [ERROR] influx - A 405 Method Not Allowed error occurred: <html>
<head><title>405 Not Allowed</title></head>
<body>
<center><h1>405 Not Allowed</h1></center>
<hr><center>nginx/1.16.1</center>
</body>
</html>

[2020-04-17T14:58:57.654] [ERROR] buffer - Failed to write points from membuffer. { error:
   'A 405 Method Not Allowed error occurred: <html>\r\n<head><title>405 Not Allowed</title></head>\r\n<body>\r\n<center><h1>405 Not Allowed</h1></center>\r\n<hr><center>nginx/1.16.1</center>\r\n</body>\r\n</html>\r\n' }
[2020-04-17T14:58:57.656] [ERROR] buffer - Will try again in 5s.

coussej, 2020-04-17:
Is it possible you are running a beta of influxDB 2.0?

JanMaasland, 2020-04-17:
Yes. I'm using influxDB 2.0. Is it not tested with the OPC-UA logger?

coussej, 2020-04-18:
It has a different API than influx 1.x. We have not yet included support for influx 2.0, but will do soon :-)

maziarz, 2020-04-24:
I have the same problem here. Authentication method does not seem to work with this logger, as it needs to use a token.

coussej, 2020-04-27:
@maziarz InfluxDB 2.0 is not yet supported. We'll implement it soon!

coussej, 2020-05-18:
Hello @maziarz, you can find initial support for 2.0 on the dev branch! 

when you add a token, org and bucket to the config file, the logger will assume its influx 2.0 and connect as such.

closed 2020-10-12 by coussej