coussej/node-opcua-logger #47: cannot start the Win-64 service

jasonye87, 2020-05-27:
Hello Sir,

I tried to run the Win-64 .exe service and it throws out an error pointing to 
node-opcua-logger\node-modules\toml\lib\parser.js:3833 as shown in the capture.
[image]

Could you help check the error and share the solution?

Thanks,
Jason

coussej, 2020-05-27:
Passing the config file with a flag won't work, if you want to use a custom location you should set it in the Environment (see also https://superuser.com/questions/79612/setting-and-getting-windows-environment-variables-from-the-command-prompt)

However, `config.toml` is the default location for the config file, and from the syntax error it looks like it's not formatted correctly. The error specifies it found a `{`, did you perhaps use JSON instead of TOML? 

Can you share the contents of the file here?

jasonye87, 2020-05-27:
Thanks for the reply, Jeroen.
After changing the config.toml file along with the sample file, now the application works well.

Just curious about how to set and use config.json file instead?

Regards,
Jason

coussej, 2020-05-27:
Hello, 

the application determines the format based on the extension of the config file. You would set the necessary environment variable by running the following command:
```
set CONFIG_FILE=c:\path\to\config.json
```
and then in the same command line window you run the binary of the logger! 

Usually, after testing config etc, you'd run the logger through a windows service. I usually suggest using [nssm](https://nssm.cc/) to do so. During the installation, you can also specify the environment variables for the service.

jasonye87, 2020-05-28:
Got it, thanks for the help!

coussej, 2020-05-28:
You'r welcome!

closed 2020-05-28 by coussej