coussej/node-opcua-logger #48: Server Rejects Client

ajaywazir, 2020-05-27:
Hi 
I am using Matrikon OPC simulation server to test the opc client
Here is the console log message

`[2020-05-27T16:35:40.091] [INFO] main - Starting Influx OPCUA Logger v2.0.0-alpha, brought to you by FACTRY (www.factry.io)
[2020-05-27T16:35:40.094] [INFO] main - Initialising influxClient
[2020-05-27T16:35:40.111] [INFO] influx - localhost:8086 responded in 14ms running 1.7.2
[2020-05-27T16:35:40.112] [INFO] main - Initialising buffer
[2020-05-27T16:35:40.116] [INFO] main - Connecting OPCUA
[2020-05-27T16:35:47.144] [FATAL] main - Error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (ERROR in waiting for data on socket ( timeout was = 60000 ms))
    at Object.start (C:\TestProjects\NodeTests\OPC\SRC\node-opcua-logger-master\src\opcua.js:64:15)
[2020-05-27T16:35:47.147] [INFO] buffer - Persisting membuffer.`

Why is certificate rejected by the server?

Ajay

coussej, 2020-05-27:
Hi there, you must trust the certificate from the server side. You'll probably find a menu in the matricon OPCUA server configuration where you can select this certificate and trust it.

ajaywazir, 2020-05-27:
Thanks. Let me check.
If there is no option, how to change to connection without certificates?
Thanks
Ajay

coussej, 2020-05-27:
This also depends on your server settings I believe. You'll find a lot of settings related to authentication and so on in the endpoint configuration.

ajaywazir, 2020-05-27:
Thanks
Let me try...
Ajay

ajaywazir, 2020-05-28:
Hi
I Got it Working
Thanks
Ajay

closed 2020-05-28 by ajaywazir