coussej/node-opcua-logger #50: Error: The connection has rejected by server

nexun, 2020-06-09:
Hi, I am having a problem running the npm start command. It connects seamlessly to the database but when it wants to connect to the OPC UA it throws an error. I have configured the simulator and I am accessing through the url that provides me. 
How should I do to configure the certificate correctly?

the opc ua server url is generated by the prosys opc ua server simulador

`# The OPCUA connection parameters. If you want to use anonymous auth, 
# remove the username and password lines.
[opcua]
url= "opc.tcp://c******:53530/OPCUA/SimulationServer"

# The InfluxDB connection parameters. Use a connection url containing all 
# details, ie. http(s)://user:password@host:port/database
[influx]
url              = "http://myuser:12345@192.168.1.10:8086/OPC_DB"
writeInterval    = 1000          # optional. defaults to 1000ms
writeMaxPoints   = 1000          # optional. defaults to 1000 points

# For each metrics you want to collect, add a [[metrics]] object.
[[metrics]]
measurement        = "polled1"
datatype           = "number"
tags               = { simulation = "true", location = "ghent" }
nodeId             = "ns=5;s=Sinusoid1"
method             = "polled"
interval           = 1000     

[[metrics]]
measurement        = "monitored2"
datatype           = "boolean"
tags               = { simulation = "true", location = "ghent" }
nodeId             = "ns=3;s=BooleanDataItem"
method             = "monitored"
interval           = 5000  
`

Thanks

nexun, 2020-06-10:
Fix the error by installing version 2

closed 2020-06-10 by nexun

JanRadtke, 2022-03-26:
Which version 2?