coussej/node-opcua-logger #52: Unable to parse

azwald, 2020-07-02:
Hi,
I got this issue from alpha 2.0 build
```
 {"error":"unable to parse 'Oil 1 Real,equipment=Mixer\\ 1,status=Good value=12.5 1593421231055000000': invalid field format
unable to parse 'Cool Mix Control Temp,equipment=Mixer\\ 1,status=Good value=464 1593421231040000000': invalid field format
```

and this is the config for that metric
```
[[metrics]]
measurement = "Oil 1 Real"
dataType    = "number"
tags        = { equipment = "Mixer 1" }
nodeId      = "ns=2;s=_AdvancedTags.Oil Group.Oil 1 Real"
method      = "polled"
interval    = 300000

```
basically, all metrics are failed from being written to influxdb, any pointers for help?

coussej, 2020-07-02:
First thought: could you try without spaces in the measurement name? If it
works like that, it's a bug in our code with spaces in the name, and we
should fix it :-)

azwald, 2020-07-02:
tried changing the space, and it works!
but I think the old data in buffer keep trying to be written to db, how to clear the buffer? thanks

coussej, 2020-07-02:
Just clear the buffer files in the buffer folder that was created. :-)

azwald, 2020-07-02:
thanks for the help! for now it seems I'd need to rename all measurements

closed 2020-10-12 by coussej