coussej/node-opcua-logger #55: OPC UA Server Close

Didier2401, 2020-10-01:
Hi Jeroen,

First of all thanks again for your really nice app. Its working perfectly but I have a small problem.
I want to know if its possible to do something in case of the OPC UA server is restarting due to an update of software in PLC.
Like a loop to retry many times a reconnection with the OPC UA server and not to close the app like it is actually.

Thanks a lot

Didier

coussej, 2020-10-01:
Hi Didier, 

I tend to solve this by running the app as a server, so it automatically restarts in case such an event occurs. Are you running on windows or linux? On windows I suggest using nssm (https://www.nssm.cc/download), on linux systemd.

Didier2401, 2020-10-02:
Hi Jeroen,

Thanks a lot for your quick answer. 
I'm working with linux. (Ubuntu 18.04). Do you have an example of service file for systemd to restart the logger automatically ?

Thanks a lot and have a nice weekend

Didier

coussej, 2020-10-03:
you could use something like:
```
[Unit]
Description=node-opcua-logger

[Service]
Environment=CONFIG_FILE=/etc/node-opcua-logger/my-config.toml
Environment=DATA_PATH=/var/opt/node-opcua-logger/
Type=simple
ExecStart=/opt/node-opcua-logger/influx-opcua-logger_v2.0.0-alpha_linux-x64
Restart=always
RestartSec=10
User=your-user
Group=your-user-group

[Install]
WantedBy=multi-user.target
```
Of course, adapt it to the paths and user you use on your system.

Does that work for you?

Didier2401, 2020-10-05:
Hi Jeroen,

Thanks a lot it help me a lot.

Didier

closed 2020-10-05 by Didier2401