coussej/node-opcua-logger #56: Unauthorized error

LukasTelvak, 2020-10-12:
Hello, I am trying to send data from OPCUA Server to InfluxDB 2.0 using your plugin, but I am getting an unauthorized error. First I thought that I used wrong token, which didn't have needed permissions but I checked that and I don't see any problem here. So I would be really thankfull if you could help me.

Here is log containing error message
[2020-10-12T13:28:13.747] [INFO] main - Starting Influx OPCUA Logger v2.0.0-alpha, brought to you by FACTRY (www.factry.io)
[2020-10-12T13:28:13.750] [INFO] main - Initialising influxClient
[2020-10-12T13:28:13.822] [INFO] influx - 192.168.0.25:8086 responded in 63ms running
[2020-10-12T13:28:13.832] [INFO] main - Initialising buffer
[2020-10-12T13:28:13.838] [INFO] main - Connecting OPCUA
[2020-10-12T13:28:13.842] [INFO] buffer - Start processing of bufferfile { file: '/home/data/_queue/1602493702208_0.json' }
[2020-10-12T13:28:13.866] [ERROR] influx - A 401 Unauthorized error occurred: {"code":"unauthorized","message":"unauthorized access"}
[2020-10-12T13:28:13.867] [ERROR] buffer - Error writing points { error:
   'A 401 Unauthorized error occurred: {"code":"unauthorized","message":"unauthorized access"}' }
[2020-10-12T13:22:01.403] [INFO] opcuaclient - Established connection. { Endpoint: 'opc.tcp://192.168.0.21:12345' }
[2020-10-12T13:22:01.484] [INFO] opcuaclient - Established session with server. { ID: 26 }
[2020-10-12T13:22:01.486] [INFO] opcuaclient - Added Polled Metric. { measurement: 'U-L1L2[V]',
  tags: { simulation: 'true', location: 'Levice' },
  nodeId: 'ns=4;s=Sensor_Simulator:InputRegisters:0:0' }
[2020-10-12T13:22:01.486] [INFO] opcuaclient - Added Polled Metric. { measurement: 'U-L2L3[V]',
  tags: { simulation: 'true', location: 'Levice' },
  nodeId: 'ns=4;s=Sensor_Simulator:InputRegisters:1:0' }
[2020-10-12T13:22:01.487] [INFO] opcuaclient - Added Polled Metric. { measurement: 'U-L3L1[V]',
  tags: { simulation: 'true', location: 'Sala' },
  nodeId: 'ns=4;s=Sensor_Simulator:InputRegisters:2:0' }
[2020-10-12T13:22:01.520] [INFO] opcuaclient - Installed subscription on session. { ID: 84 }
[2020-10-12T13:22:02.397] [INFO] buffer - Start processing of bufferfile { file: '/home/data/_queue/1602493702208_0.json' }
[2020-10-12T13:22:02.403] [ERROR] influx - A 401 Unauthorized error occurred: {"code":"unauthorized","message":"unauthorized access"}
[2020-10-12T13:22:02.404] [ERROR] buffer - Error writing points { error:
   'A 401 Unauthorized error occurred: {"code":"unauthorized","message":"unauthorized access"}' }

And here is my config file
[opcua]
url = "opc.tcp://192.168.0.21:12345"

[influx]
url = "http://lukas:password@192.168.0.25:8086/snmpbucket"
token = "oI67t9BXnHnoYtWRl7YBen9pbZGQngrFje39gBr1WC8TMPk46gclqPFGOS3OK_-eb84EUOg_zRPJth74hp1Paw=="
org = "myorg"
bucket = "snmpbucket"
writeInterval = 10000

coussej, 2020-10-12:
That seems to be an error coming from the influx client library itself. Are you certain the token is valid, and provides access to the bucket you want to write to?

LukasTelvak, 2020-10-13:
Thanks for the reply, yes I tried creating two new tokens one with access just to the bucket I want to write and when that didn't work, I tried using one that has access to everything in Influx and I am still getting same error and nothing is sent to database.

coussej, 2020-11-24:
Can you access the database from grafana or chronograph using those credentials?

vkhemani, 2021-05-03:
hi, i am still stuck at this problem. i am getting the same errors. With the same credentials in the config file, i am able to access database in Grafana. but, unable to send data to the database from the logger.