coussej/node-opcua-logger #57: Connection reject by server

barthelemyguillaume, 2020-10-26:
Hi,

I've an error when I connect to my OPC Server.

`Error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (connect ETIMEDOUT xxx.xxx.xxx.xxx:4840)
    at Object.start (/snapshot/node-opcua-logger/src/opcua.js:0:0)`

I use anonymous auth.

I try with UA Expert Client and it's ok.

What's the certificate?

Regards

mirkocomparetti-synesis, 2020-11-02:
I'm having a similar issue with anonymous auth and I'm getting back an `[FATAL] main - Error: Invalid server Nonce` which should not be there as that check has to be skipped in case of anonymous auth.
All other OPC-UA clients are working fine.

coussej, 2020-11-03:
Even with anonymous auth sometimes the server requires the certificate to be accepted. With UA server are you using?

mirkocomparetti-synesis, 2020-11-03:
Hello,
I'm using a server from a Schneider Electric PLC. UA expert and a lot of other clients are able to connect flawlessly but the only one that is reporting the invalid nonce is this one, which we like and need so badly ;)
Anyway I also opened a ticket to the node-opcua library https://github.com/node-opcua/node-opcua/issues/904#issuecomment-720960288

Thanks!

Rifasca, 2020-11-04:
Hi,
I am also facing similar issue when I try to run the code. Need help. But when I tried running influx-opcua-logger_v2.0.0-alpha_win-x64.exe, its working fine and I am getting the response.
[2020-11-04T12:06:36.561] [INFO] main - Starting Influx OPCUA Logger v2.0.0-alpha, brought to you by FACTRY (www.factry.io)
[2020-11-04T12:06:36.562] [INFO] main - Initialising influxClient
[2020-11-04T12:06:36.571] [INFO] influx - localhost:8086 responded in 6ms running 1.7.6
[2020-11-04T12:06:36.572] [INFO] main - Initialising buffer
[2020-11-04T12:06:36.574] [INFO] main - Connecting OPCUA
[2020-11-04T12:06:36.604] [FATAL] main - Error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Internal Error)
at Object.start (D:\SPIA\opcua\node-opcua-logger-2.0.0-alpha\src\opcua.js:60:11)
[2020-11-04T12:06:36.605] [INFO] buffer - Persisting membuffer.

mirkocomparetti-synesis, 2020-11-04:
> Hello,
> I'm using a server from a Schneider Electric PLC. UA expert and a lot of other clients are able to connect flawlessly but the only one that is reporting the invalid nonce is this one, which we like and need so badly ;)
> Anyway I also opened a ticket to the node-opcua library [node-opcua/node-opcua#904 (comment)](https://github.com/node-opcua/node-opcua/issues/904#issuecomment-720960288)
> 
> Thanks!

As regards my request, this has been fixed by  [node-opcua/node-opcua#904 (comment)](https://github.com/node-opcua/node-opcua/issues/904).

Thanks

Rifasca, 2020-11-05:
> As regards my request, this has been fixed by [node-opcua/node-opcua#904 (comment)](https://github.com/node-opcua/node-opcua/issues/904#issuecomment-721689655).
>
> Thanks

Thanks for the help. The code is working fine after I upgraded the opcua client.

closed 2022-03-08 by barthelemyguillaume