coussej/node-opcua-logger #59: Rasberry PI4 test : connection rejected by server

stephb59, 2020-11-20:
Hi, Thanks for the job done on this. I would like to test this on an rpi4.
I have these certificate error while running
<pre><font color="#4E9A06"><b>pi@IOTstack</b></font>:<font color="#3465A4"><b>~/node-opcua-logger $</b></font> npm start

&gt; influx-opcua-logger@2.0.0-alpha start /home/pi/node-opcua-logger
&gt; node ./src/index.js

<font color="#4E9A06">[2020-11-20T16:07:01.422] [INFO] main - </font>Starting Influx OPCUA Logger v2.0.0-alpha, brought to you by FACTRY (www.factry.io)
<font color="#4E9A06">[2020-11-20T16:07:01.427] [INFO] main - </font>Initialising influxClient
<font color="#4E9A06">[2020-11-20T16:07:01.462] [INFO] influx - </font>localhost:8086 responded in 28ms running 1.8.3
<font color="#4E9A06">[2020-11-20T16:07:01.463] [INFO] main - </font>Initialising buffer
<font color="#4E9A06">[2020-11-20T16:07:01.466] [INFO] main - </font>Connecting OPCUA
<font color="#75507B">[2020-11-20T16:07:02.477] [FATAL] main - </font>Error: The connection has been rejected by server,
Please check that client certificate is trusted by server.
Err = (Internal Error)
    at Object.start (/home/pi/node-opcua-logger/src/opcua.js:60:11)
<font color="#4E9A06">[2020-11-20T16:07:02.481] [INFO] buffer - </font>Persisting membuffer.
</pre>
I try to connect to this opc server available online `opc.tcp://opcuaserver.com:48010`. See config.toml attached
I tried with release V2 and master compiled, i have the same error.
Prebuilt binary not working on raspberry 32 bits.
[config.toml.log](https://github.com/coussej/node-opcua-logger/files/5574389/config.toml.log)

I connected successfully uaexpert on that opc server, anonymous connection.

Can you give ideas about that problem ?
Thanks in advance

coussej, 2020-11-24:
Indeed, the pre built binaries don't work on ARM.  

Which node version did you install on the PI?

stephb59, 2020-11-24:
It is v10.23.0

stephb59, 2020-12-01:
Update.
There was a problem with openssl. I reinstalled nodejs and logger again.
Working perfectly now. 
I am able to connect the online opc server above successfully.  
FYI i used the IOTstack from https://sensorsiot.github.io/IOTstack/ which deploy influx and grafana within 10 minutes.
Great job for this project. Thanks to the team.

closed 2020-12-01 by stephb59