coussej/node-opcua-logger #62: the certificate and the private key do not match !

stephb59, 2021-03-17:
Hi,
I successfully installed my first logger on a raspberry pi previously.
New instance is on a linux VM on xenserver
Linux is
`Linux pi-OPC 5.4.0-56-generic #62-Ubuntu SMP Mon Nov 23 19:20:19 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`
Install is from source
On a new install i get  the following messages
```
pi@pi-OPC:/data/IOTstack/node-opcua-logger$ npm run start

> influx-opcua-logger@2.0.0-alpha start /data/IOTstack/node-opcua-logger
> node ./src/index.js

[2021-03-17T10:14:31.237] [INFO] main - Starting Influx OPCUA Logger v2.0.0-alpha, brought to you by FACTRY (www.factry.io)
[2021-03-17T10:14:31.239] [INFO] main - Initialising influxClient
[2021-03-17T10:14:31.255] [INFO] influx - localhost:8086 responded in 13ms running 1.8.3
[2021-03-17T10:14:31.256] [INFO] main - Initialising buffer
[2021-03-17T10:14:31.258] [INFO] main - Connecting OPCUA
09:14:31.304Z :verify                        :110   [NODE-OPCUA-E01] Configuration error : the certificate and the private key do not match !
09:14:31.306Z :verify                        :111                     please check the configuration of the OPCUA Server
09:14:31.306Z :verify                        :112                       privateKey=  /home/pi/.config/node-opcua-default-nodejs/PKI/own/private/private_key.pem
09:14:31.307Z :verify                        :113    certificateManager.privateKey=  /home/pi/.config/node-opcua-default-nodejs/PKI/own/private/private_key.pem
09:14:31.307Z :verify                        :114                  certificateFile=  /data/IOTstack/node-opcua-logger/certificates/client_selfsigned_cert.pem
[2021-03-17T10:14:31.308] [FATAL] main - Error: [NODE-OPCUA-E01] Configuration error : the certificate and the private key do not match ! please fix your configuration
    at Object.start (/data/IOTstack/node-opcua-logger/src/opcua.js:60:11)
    at async /data/IOTstack/node-opcua-logger/src/index.js:50:5
[2021-03-17T10:14:31.309] [INFO] buffer - Persisting membuffer.

```
I then tried the pre built binary `influx-opcua-logger_v2.0.0-alpha_linux-x64` which is working flawless with the same config.toml file
[config.toml.txt](https://github.com/coussej/node-opcua-logger/files/6155199/config.toml.txt)
The output is like that with prebuilt
```
pi@pi-OPC:/data/IOTstack$ ./influx-opcua-logger_v2.0.0-alpha_linux-x64
[2021-03-17T10:08:24.351] [INFO] main - Starting Influx OPCUA Logger v2.0.0-alpha, brought to you by FACTRY (www.factry.io)
[2021-03-17T10:08:24.353] [INFO] main - Initialising influxClient
[2021-03-17T10:08:24.405] [INFO] influx - localhost:8086 responded in 48ms running 1.8.3
[2021-03-17T10:08:24.418] [INFO] main - Initialising buffer
[2021-03-17T10:08:24.421] [INFO] main - Connecting OPCUA
[2021-03-17T10:08:25.240] [INFO] opcuaclient - Established connection. { Endpoint: 'opc.tcp://opcuaserver.com:48010' }
[2021-03-17T10:08:25.556] [INFO] opcuaclient - Established session with server. { ID: '2DF5345E-00DD-487A-8FD7-50F77B8B1562' }
[2021-03-17T10:08:25.557] [INFO] opcuaclient - Added Polled Metric. { measurement: 'length',
  tags: { simulation: 'true', location: 'www' },
  nodeId: 'ns=3;s=AirConditioner_1.Temperature' }
[2021-03-17T10:08:25.558] [INFO] opcuaclient - Added Monitored Metric. { measurement: 'length',
  tags: { simulation: 'true', location: 'www' },
  nodeId: 'ns=3;s=AirConditioner_1.Temperature' }
[2021-03-17T10:08:25.715] [INFO] opcuaclient - Installed subscription on session. { ID: 1978692350 }

```
I can use the prebuilt but i do not know why it is not working with the source compiled version

coussej, 2021-09-21:
Can you try regenerating the certificates with npm run postinstall before building?

stephb59, 2021-09-23:
I just make a new install from source right now and everything work. :-)

closed 2021-09-23 by stephb59