coussej/node-opcua-logger #64: Future development #question

MoritzLaut, 2021-04-08:
is this project still ongoing and are there any future plans e.g. Influx 2.0?

~Moritz

robberthofmanfm, 2021-08-23:
Hello @MoritzLaut and @coussej 

There is a telegraf plugin for OPC-UA now. It works with InfluxDBv2.
See also
https://github.com/influxdata/telegraf/blob/master/plugins/inputs/opcua/README.md
https://docs.influxdata.com/influxdb/v2.0/write-data/no-code/use-telegraf/manual-config/

henjoe, 2021-09-21:
To be honest, I've used telegraf opcua, but having so much troubles and issues. And I don't see if there is a data buffering on the side of telegraf when connection to influx is lost. 

On my opinion, opcua logger is still the best tested application. 💪 

I hope we can run multiple instances of this opcua logger on 1 computer, or atleast it has configuration for multiple opc server inputs and multiple influxdb output ☺️

stephb59, 2021-09-21:
I successfully runned 3 parallel opcua logger to a single influx db. I do not think this could be a problem to output to different db's.
I used IOTstack (https://sensorsiot.github.io/). This is grafana, influx, and more already configured within docker.

henjoe, 2021-09-21:
Hi @stephb59 how did you that? That was amazing. I never got lucked running atleast 2 insctance of opcua logger on my windows machine. 
Does it need to be run separately on a a docker? 
I am just using the opcualogger.exe for this. 
I hope you can help me.

stephb59, 2021-09-21:
I never tried with windows. I do prefer working on "the light side of the force" :-)
My hardware is a small linux server. Linux always working more efficiently.
I just do have several instances that are not on the docker side.  Grafana and Influx are within docker. Several opcua logger instances are pushing data to influxdb.

stephb59, 2021-09-21:
On my small server i did some tests recording online opcua server with 5 instances

henjoe, 2021-09-21:
@stephb59 is it via docker? (You've run 5 docker with opcua logger) Or you've just run it 1by1 on a command line? 

There is no conflict on instance-uid?

coussej, 2021-09-21:
Hello all,

- @MoritzLaut  it's true that not a lot of development has been done the past months, but support for influx 2.0 is already available in the dev branch! 

- @henjoe you should be able to run different loggers next to each other. You should just configure different paths for the config file, data en log files, and everything should be fine!

henjoe, 2021-09-21:
Hi 
@coussej never knew about this. Thanks a lot!

stephb59, 2021-09-21:
@henjoe  On main os i run it in 5 different screen sessions with different path and configuration files. No conflict as instances are different pid. Docker containers are only there for influx and grafana.

henjoe, 2021-09-21:
Hi @stephb59  and @coussej,
One more thing. May I confirm that we can run multiple instances of opcua logger when it was run from a source? Instead of a prebuilt binary?

I am trying to run multiple instances  from the pre built binary but it seems that only one is running.

Hoping for your confirmation. 
Thanks in advance! :)

coussej, 2021-09-21:
You can provide the path to the config file using environment variables, you can see which ones in the readme file. You can for example create a windows service from the same binary, but configure different parameters for each service. [nssm](https://nssm.cc/) has an easy way to do this for example.

stephb59, 2021-09-21:
In my linux case
```
node-opcua-logger1
├── config.toml
├── config.toml.old
├── data
│   ├── _error
│   └── _queue
└── influx-opcua-logger_v2.0.0-alpha_linux-x64
node-opcua-logger2
├── config.toml
├── data
│   ├── _error
│   └── _queue
└── influx-opcua-logger_v2.0.0-alpha_linux-x64
node-opcua-logger3
├── config.toml
├── data
│   ├── _error
│   └── _queue
└── influx-opcua-logger_v2.0.0-alpha_linux-x64
node-opcua-logger4
├── config.toml
├── data
│   ├── _error
│   └── _queue
└── influx-opcua-logger_v2.0.0-alpha_linux-x64
```

henjoe, 2021-09-21:
> You can provide the path to the config file using environment variables, you can see which ones in the readme file. You can for example create a windows service from the same binary, but configure [...]

Hi @coussej , I understand that I can use an environment variables for the config files. What I didn't know is how would I set other instance of opcualogger to use other config path using environment variables? Like for example if I use an environment variables foe opcualogger1, how would my opcualogger2 will use another environment variables? (with the same variable name of opcualogger1) 

Is there a paraneter like running the opcualogger --config path?

henjoe, 2021-09-21:
> In my linux case
> ```
> node-opcua-logger1
> ├── config.toml
> ├── config.toml.old
> ├── data
> │   ├── _error
> │   └── _queue
> └── influx-opcua-logger_v2.0.0-alpha_linux-x64
> node-opcua-logger2
> ├── config.toml [...]

Hi @stephb59 have you tried this running

stephb59, 2021-09-22:
@henjoe 
There is no issue sending data to same port. Influx db like others db's can accept simultaneous connections. I did not try with different databases.

henjoe, 2021-09-22:
@stephb59 

I found the issue now. May testing machine is blocking the opcua program 🤷 

I switch to another machine and I was succesfull running it simultaneously. 

Again, thanks for your help! :)

closed 2021-12-14 by MoritzLaut