coussej/node-opcua-logger #70: [Question] Adding OPCUA Tag while running

henjoe, 2021-06-04:
Hi Coussej, 

First of all, I would like to commend you for this great project! Indeed your ideas of combining OT-IT as the future trend was so amazing.  

In anyways, I would like to ask if it is possible that this tool, while it is running can we add a Tags on it? like simply adding a [[measurement]] on a config.toml. Does it needs to be restarted before the added [[measurements]] take effects? I am just thinking that if we restarted it, there will be a lost of data.

Again. Thank you on your greate work!

Regards,
Henjoe

coussej, 2021-09-21:
Yes, in this open source implementation the application has to be restarted to reload the config file. The commercial version we've developed with Factry has an advanced management UI where tags can be added and modified on the fly, without interrupting the others.

closed 2021-09-21 by coussej

coussej, 2021-09-21:
You can use one of the contact methods on https://www.factry.io/ and refer to this conversation, it will end up with me then ;-) thanks!