coussej/node-opcua-logger #71: [Question] Clarification on the side of BatchSize, buffer.js

henjoe, 2021-06-24:
Hi Coussej,

I am wondering what is the meaning of this configuration on the side of Buffer.js 
[image]

Does it mean only 1000 Tags are going to buff? 
I received warnings that may memBuffer already exceed a maximum of 1000. 
Is is something to do with the batchSize right? 

Thank you,

coussej, 2021-09-21:
Hi @henjoe,

That simply means the maximum batch size that will be written to influx. Let's say you have 2000 metrics per second, they will be written to influx in 2 batches of 1000 points. You can increase that if you have more.

It is not related to the amount of tags that are going to be buffered.

Regards,
Jeroen

closed 2021-09-21 by coussej

henjoe, 2021-09-21:
Hi @coussej 
Thanks for this! :) 

Higly appreciated your work here.